markusdd/OpenDTUFusionDocs #2: Radio modules not recognized after OTA update

0per8or, 2023-07-05:
I recently bought a OpenDTU Fusion board. After trying to do a OTA update to the latest version it seems that the radio modules do not get recognized correctly.

[image]
[image]

markusdd, 2023-07-05:
Hi. This can happen as the boards are delivered with a firmware that has the pins baked in.
If you update with the generic image you might need to load the opendtufusion specific device profile jason which has all the pin mapping options (also for displays) as described here: https://github.com/markusdd/OpenDTUFusionDocs/blob/main/FLASHED_START.md#loading-a-device-profile-for-the-rf-and-display-pinout

If you keep using the generic OTA builds in the future you should not need to do this again.

0per8or, 2023-07-05:
This fixed it for me. Thank you!

Maybe it is a good idea to put this kind of information to the Readme to prevent others from running into the same issue.

closed 2023-07-05 by 0per8or

markusdd, 2023-07-05:
Done: https://github.com/markusdd/OpenDTUFusionDocs/blob/main/OTA.md#attention-for-pre-flashed-boards