markusdd/OpenDTUFusionDocs #3: no success when flashing with AhoyDTU

341bms, 2023-07-13:
Has anyone already had experience flashing with AhoyDTU? It did not work for me, at least via the [web installer](https://ahoydtu.de/web_install/). The process could never be initiated. The connection via USB to the board basically works.

markusdd, 2023-07-13:
The web flasher is not compatible with the new ESP32-S3 that the board uses.
So you either have to use VS Code or the Flash Tool. (or esptool on command line if you are familiar)

For ahoy I recommend to maybe go to the Discord community. Apparently newer versions (0.7+) have issues, but I know for sure people are operating 0.6x ahoy installations on the Fusion PCB and it is generally supported.

markusdd, 2023-07-13:
yes, I made a contribution a few months back for these builds.
These do work, but must be flashed using the flash tool and not the web flasher, which is unfortunately not compatible with the ESP32-S3.

341bms, 2023-07-14:
Tried it yesterday with esptool in the command line on a Mac.... but the AP never came up...

markusdd, 2023-07-14:
did you do a complete erase before flashing?
which image did you use?

341bms, 2023-07-14:
I use esptool.py v4.6.2... 

Doing the erase:

```
% esptool.py --port /dev/cu.usbmodem101 erase_flash 
esptool.py v4.6.2
Serial port /dev/cu.usbmodem101
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:67:18:24
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 27.6s
Hard resetting via RTS pin...
```
Flashing with esp32s3.bin:

```
% esptool.py --port /dev/cu.usbmodem101 write_flash 0x0 ~/Downloads/ahoydtu_dev/firmware/s3/230709_ahoy_0.7.11_0be3a23_esp32s3.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem101
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:67:18:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00110fff...
Compressed 1117728 bytes to 718978...
Wrote 1117728 bytes (718978 compressed) at 0x00000000 in 9.5 seconds (effective 942.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
The device then seems to be in a continuous loop and does not boot properly:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379823
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379835
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037982f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379826
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
```

I also played around with the `partitions.bin` and `bootloader.bin`. But only on suspicion because I could not find any specific instructions.

markusdd, 2023-07-14:
it clearly complains about a SHA mismatch of the flashed image.

Are you using the partitions.bin and bootloader from ahoy? The bootloader I think is somewhat fixed out of the framework but the partition tables are incompatible.
Is has to be something like this because otherwise everything is working as intended from the logs. But when it finally tries to boot the image it gets garbage.

341bms, 2023-07-14:
partitions.bin and bootloader.bin that I tried once are from ahoy, right. 

Just tried again with the 7.0.10 dev build. Checksum does not fit here either.

```
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc644
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: a51fcdb96f8d08f6828a44cc9096ee9493bc29c0774da8dc785d0aa7f9c42879
Expected: 73ff2a718eaa2957f53110280070691162df96cdaa72b4e6140765e4fd464117
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc644
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: a51fcdb96f8d08f6828a44cc9096ee9493bc29c0774da8dc785d0aa7f9c42879
Expected: 73ff2a718eaa2957f53110280070691162df96cdaa72b4e6140765e4fd464117
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379837
SPIWP:0xee
mode:DIO, clock div:1
```

markusdd, 2023-07-14:
wow. This is super weird.

I do not believe this to be the cause but just for good measure can you try another USB cable?

341bms, 2023-07-14:
Same same with a different USB cable.

```
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379837
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
Attempting to boot anyway...
entry 0x40377410
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c3
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c0c0020,len:0x3c5ec
load:0x3fc962b0,len:0x3a04
load:0x42000020,len:0xbc640
load:0x3fc99cb4,len:0x24d8
load:0x40374000,len:0x122ac
SHA-256 comparison failed:
Calculated: b0cfd969aebe52e470e4f2f73512c21d4f6a82f2c1e97e0e58299f376f155d0c
Expected: f7d4f78b4f44cb49897eca2ca7f785031f13c4549424103f74cea578c53ca8c9
```

To be sure I'm doing the right thinks. Just erasing and flashing with the s3 dev bin without pressing bottoms on the pcb or flashing other bins, right?

markusdd, 2023-07-14:
I just see in the log you are trying to flash the firmware from address 0x0.

I think for ahoy this is not correct as they do not use a completely merged image like OpenDTU does.

Can you crosscheck their repo README for the addresses to use?

markusdd, 2023-07-14:
Firmware goes at 0x10000

https://github.com/lumapu/ahoy/blob/development03/Getting_Started.md#flashing-on-linux-with-esptoolpy-esp32

So the call should be `esptool.py --port /dev/ttyUSB0   write_flash --flash_mode dio --flash_freq 40m --flash_size detect   0x1000 bootloader.bin   0x8000 partitions.bin   0x10000 esp32.bin`

341bms, 2023-07-14:
Tried again with this command: `esptool.py --port /dev/cu.usbmodem101 --before default_reset --after hard_reset write_flash --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x8000 partitions.bin 0x10000 230709_ahoy_0.7.11_0be3a23_esp32s3.bin`

```
esptool.py v4.6.2
Serial port /dev/cu.usbmodem101
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:67:18:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00001000 to 0x00004fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00120fff...
Compressed 15040 bytes to 10333...
Wrote 15040 bytes (10333 compressed) at 0x00001000 in 0.3 seconds (effective 403.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 376.5 kbit/s)...
Hash of data verified.
Compressed 1117728 bytes to 718978...
Wrote 1117728 bytes (718978 compressed) at 0x00010000 in 9.4 seconds (effective 948.4 kbit/s)...
Hash of data verified.

```

Now getting logs like this:

```
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

:/

markusdd, 2023-07-14:
ok xD

I really do not know ahoy well enough it seems, I really recommend to go to the discord community into the ahoy support channel, there will be more knowledgable people to help out

this is certainly just a small process issue.

markusdd, 2023-08-31:
as there have been some updates regarding ahoy in the past weeks I am closing here, anything more should be discussed on the comunity dicord if needed.

closed 2023-08-31 by markusdd

341bms, 2023-09-08:
Flashing works now. Previously the address for the bootloader was not correct. It must be **0x0000** instead of 0x1000. So it works like this: 

`esptool.py --port /dev/cu.usbmodem1101 --before default_reset --after hard_reset write_flash --flash_mode dout --flash_freq 40m --flash_size detect 0x0000 bootloader.bin 0x8000 partitions.bin 0x10000 230907_ahoy_0.7.47_3aa9947_esp32s3.bin`