markusdd/OpenDTUFusionDocs #4: Pin mapping for OpenDTU-onBattery

swingstate, 2023-08-27:
Hi Markus, 

two pre-sales question - 

I have read the docs and understand that some of the GPIOs (9 - 14) can be mapped to anything else. I would like to map the VE.Direct feature of OpenDTU-onBattery to e.g. GPIO 9 and 10 from the SPI header. Would that work?

Are the IPEX connectors standard size so other pigtails for SMA antennas can be connected?

markusdd, 2023-08-27:
yes.  They are normal IPEX connectors. (as indicated in the docs, not the 'mini' ones you find on some newer laptop wifi cards).

As for re-using GPIOs: they are the normal 3.3V GPIOs of the ESP32-S3 then. I do not know enough about Victron if 3.3V is the correct voltage for VE.Direct, that would need to be asked at OpenDTO onBattery.

We do nothing on the board with these lines except routing them to the pin-header straight form the ESP32-S3 module. So there are no pulls/level shifters etc.

philippsandhaus, 2023-08-28:
Regarding VE.Direct: There are some Victron devices like the MPPT Solar chargers which use 5V and some use 3.3 V like the SmartShunt. I also plan to use the headers on the Fusion Board for this, but I plan to solder a header socket on a ADUM1201 level shifter with a VE. Direct cable or socket on the other side and place it directly on the 4header group with the SDA/SCL Pins of the board. If this works it should a nice and clean solution.

swingstate, 2023-08-28:
Thanks Markus, 

The VE.Direct implementation for OpenDTU makes use of the GND, TX, RX Pins. The VCC PIN remains unused. Hence my understanding is that voltages are not a topic for this interface. However, does any of the available PINs on the fusion board provide GND?

markusdd, 2023-08-28:
That is incorrect. Of course the high and low levels of the signals matter. If you drive a 5V signal into an ESP32 you will destroy it. So anything you connect to an ESP board (not just the fusion) must not exceed that 3.3V voltage on the signal pins.
GND is not an issue, it is marked on the board and available at multiple pin header locations.

swingstate, 2023-08-28:
Thanks, I should have looked more closely at the pins. GND all over :). I'll give it a try and share my impressions on the OpenDTU-onbattery git.

markusdd, 2023-08-31:
I think this has been clarified, closing.

closed 2023-08-31 by markusdd

haegar33, 2024-10-05:
Catching up on this old thread about connecting a Victron MPTT. Can you please be more specific which pins to use?
I assume its the UART Pins on top (labeled UTX, URX). What is the pin_mapping layout which needs to be upload in order 
to active the pins?

Thanks, Haegar

markusdd, 2024-10-05:
Please read the documentation about the CAN/Iso shield. The Victron cannot be connected to the board directly without the Isolator shield, the voltages are incomptible. If you connect it directly you will destroy the ESP.

haegar33, 2024-10-05:
I am well aware about the voltage differences and use the ADUM2021 converter as mentioned in this thread. The missing info is the proper pin on the Fusionboard to use and the needed pin_layout.json file?

markusdd, 2024-10-05:
Then again: Read the documentation is all I can say. It literally has a mapping file in it.
Also: The documentation also says the S3 can literally map everything almost anywhere. So map it whereever you want, or map it where the CAN/Iso shield demands it if you rare using it.

It's literally all there.

haegar33, 2024-10-06:
I guess it would have been a good idea to use the CAN/Iso Shield from the very beginning instead of fiddling my own cables and voltage converter.  Pin Layouts and Labels can be really confusing when you have to read documentation from Espressif, OpenDTU, OpenDTUonBattery and your Fusionboard layout.

Cheers, Haegar.