markusdd/OpenDTUFusionDocs #5: OpenPDU Build Instructions

unaiur, 2023-08-31:
It would be great to have build instructions in order to build OpenPDU project with pin mappings already correctly configured.

I still did not receive the Fusion Community board (I ordered it two days ago and still in preparation stage), but I would like to start preparing the scripts...

Which variant of ESP32-S3-WROOM-1U module does it use? 

I need it to properly configure flash size and enable or disable PSRAM...

markusdd, 2023-08-31:
1) The build instruction and pin mapping JSON required are in the OpenDTU repository. This is just the hardware repository. Also I striongly advise against baking in the pins in C as this will get erased if you use the official OTA images. Use the pin_mapping.json
2) This is not the shop support and I also do not sell/ship the boards myself. If there is an issue contact them please.
3) The module used depends on availability and it is not constant. We usually aim to always at least order 8M of Flash and usually also 2M of PSRAM. But OpenDTU never assumes more than 4M of flash anyway, so there is no need to intervene there, use the standard settings.

unaiur, 2023-08-31:
Thanks for the fast answer! I found the board definition in OpenPDU and it is built for 8MiB of flash and no PSRAM. I have had a lot of issues with PSRAM in ESP32S2 boards, so I think it is better this way.

unaiur, 2023-08-31:
Also, build instructions are at BLANK_START.md

closed 2023-08-31 by unaiur

markusdd, 2023-08-31:
Yes PSRAM is also not used so this does not matter. Just leave it off.

And yes I have some basic build info in the READMEs but going to the official source is better.

I anyway recommend to use the prebuilt ones unless you are hacking yourself.