markusdd/OpenDTUFusionDocs #7: 5V Pin

l0rdnc, 2023-10-24:
Hi, I'm missing the 5V Pin in the overview png. Does the board not provide the 5V?

markusdd, 2023-10-24:
the 5V that go to the 3.3V linear regulator are either on the left or right side of the 3-pin jumper. The jumper then routes it to the center pin form where it goes to the 3.3V regulator.
[image]

In case you wish to supply 5V from USB und then also use it for something else you can just jumper all the 3 pins together and use the remaining open pin to grab the 5V from there or use a stackable jumper so you can connect whatever you need.
The reason we have not made it available on another pin before the 3.3V is that USB has relatively strict maximum capacitance requirements and some power supplies might fail supply negotiation when too much stuff is connected to 5V without a regulator in between.

markusdd, 2023-12-12:
clarified.

closed 2023-12-12 by markusdd