markusdd/OpenDTUFusionDocs #9: Canadian buyer try to buy

bebrouss, 2024-04-08:
Hi , I like to buy your board with the case on your web site, so it's not possible to me beacause Canada are not listed. SO how I can buy  the board with the case.

Thanks
you can contact me at bebrouss@hotmail.com

d-allin, 2024-06-25:
Fellow canuck here. Not really sure if asking this in the bug report section of their documentation repository is kosher, but I'll chime in.
After much faffing about on their online store (which says they ship to Canada, but then won't let you go through checkout), I ended up buying one from their ebay store (https://www.ebay.ca/usr/allianceapps).
Arrived in a week or so. No issues.

markusdd, 2024-06-25:
Hi guys, I'm indeed not running the sales operations. But @cat24max does. Tim, is it known that outside EU sales are not possible in the allianceapps store?

closed 2024-08-25 by markusdd