markusdd/OpenDTUFusionDocs #10: pin_mapping.json

Adminius, 2024-05-08:
Hi, ich versuche pin_mapping für Fusion Board mit einem SDM-Zähler über RS485 + JK-BMS + 1x Victron einzurichten.

Ich bin total verwirrt was die Pinbeschreibung angeht.
Da steht
https://github.com/markusdd/OpenDTUFusionDocs?tab=readme-ov-file#pin-mapping-of-the-esp32-s3-on-opendtu-fusion-v2x
dass Pins 22 und 23 nicht genutzt werden können, da nur Intern.
Hier werden die aber beschrieben:
https://github.com/helgeerbe/OpenDTU-OnBattery/wiki/Device-Profiles#structure-of-the-pin_mappingjson-file
Fehler?

Nun zum eigentlichen Problem: für SDM PowerMeter brauche ich RS485, und entweder nimmt man RX+TX (16+45) oder man nimmt RX+TX+DERE aber ich habe noch nie RX+TX+DE+RE gesehen. 
OpenDTU kann es auch nicht: https://github.com/helgeerbe/OpenDTU-OnBattery/blob/6620ab487a898c102158351941450ac0c8a6fbd6/src/PinMapping.cpp#L227

Vorschläge? 
Pins DE+RE zummen löten und einen Trace trennen damit es zu keinem Kurzschluss kommt? Eigentlich dachte, ich dass das Board komplett kompatibel ist :/

markusdd, 2024-05-08:
1) Wir verwalten nichts von dem was bei OpenDTUonBattery dokumentiert wird und ich kann dir dagen dass das Beispiel dort 100% falsch ist, diese Pins werden niemals funktionieren wie sie da stehen.
2) RS485 wurde auf wunsch der community damals integriert, hat aber im offiziellen OpenDTU build keinen Software-support. Es gibt aber etliche Leute die OpenDTUonBattery mit einem JK-BMS und dem Fusion nutzen.
Hier wäre es ratsam sich einmal in der discussion-section oder auf dem community.discord hinweise zu holen.
3) ich weiß nciht was du mit DE/RE willst, aber das sind die controlpins die vom ESP zum RS485 gehen, die TX/RX-Leitungen dann die zum eigentlichen Bus. Wenn du da irgendwas auftrennst kurzschließt ist der einzige effekt das der Baustein nciht mehr angesprochen werden kann.
Den Stromzähler direkt anzuschließen ist technisch möglich, braucht aber sofwtare-support. Den gibt es, oder gibt es eben nicht. Das ist mit dem jeweiligen Projektcommunities zu klären und ncihts was wir auf Hardwareebene leisten können.

Was Victron/CAN und andere Dinge angeht: es wird in Kürze wie hier verlinkt https://github.com/helgeerbe/OpenDTU-OnBattery/discussions/955 ein shield für das fusion geben um zwei isolierte seriellkanäe (z.B. VE.Direct von Victron) sowie einen CAN-Kanal (z.B. Pylontech) anzuschließen. Letzeres sowie einen VE-Direkt nutze ich bereits in einem eigenen Setup und geht wunderbar.

Sowie ich das überblicke ist dein Vorhaben mit dem Stromzähler softwareseitig entweder experimentell oder noch garnicht supported, d.h. da wird ein wenig community-engagement deinerseits notwendig sein oder zumindest ein nachhaken ob das schon jemand tut. Ich weiß nur definitiv das Leute JK-BMS erfolgreich am Fusion betreiben.

Adminius, 2024-05-08:
Über JK BMS und Victron mache ich mir keine Sorgen.

SDM Stromzähler Support habe ich vor Ewigkeiten eingebaut und Standard Aliexpess Bord mit nur RX/TX Pins genutzt.
Es gib noch alternative Module mit DE+RE Pins, diese werden aber extern immer verbunden und in allen mir bekannten Bibliotheken nur ein PIN für DE und RE gleichzeitig genutzt (oft als DERE Pin bezeichnet). Und das geht mit einem Pin in openDTU schon. Mein Problem ist, dass Fusion board mit 2 Pins arbeiten muss, weil beide an 2 GPIOs hängen. Ich kann die also nicht einfach so kurzschließen. Deswegen auch die Frage wie es angedacht war? Bzw wo am besten Cut machen damit nur mit einem GPIO verbunden bleibt?

Ich will halt von meinem Lochraster Board mit Aliexpress RS485 Module für SDM Zähler, das seit einem Jahr provisorisch hängt aber funktioniert, weg gehen, auch mit Hinblick auf 8MB Speicher.

Isolated Shild interessiert mich auch sehr, genau das, was ich brauche für Victron und JK BMS. Wobei beim JK muss man noch Step Up von 24-48V auf 5V anklemmen...

markusdd, 2024-05-08:
Wie soll es gedacht sein? Die pins sind seperat kontrollierbar, also warum dann nicht genau das tun?
Der Grund warum sie an 2 gpios gehen ist eben damit volle Kontrolle da ist, der Rest ist software.

markusdd, 2024-05-08:
That being said: so wie re/de funktionieren sollte es auch gehen einfach einen der beiden dauerhaft auf GND zu treiben und nur den anderen zu kontrollieren.

Es sollte keinesfalls nötig sein Dinge zu durchtrennen oder kurzzuschließen

Solar320, 2024-06-11:
Hmmm, Das Menu für den SDM120 ist im Open DTU on Battery vorhanden und es wird von MODBus gesprochen.

Ich verwende derzeit eine "SDM live Power Table" 

https://github.com/reaper7/SDM_Energy_Meter/blob/master/examples/sdm_live_page_esp8266_hwserial/index_page.h

Da habe ich einen ESP8266 über einen RS485/Seriell-TTL wandler an einer seriellen Schnittstelle des ESP8266 gekoppelt.

Ich gehe davon aus das das bei der DTU genauso gemacht wird. Ich vermute das hier irgend eine der seriellen Schnittstellen
verwendet werden kann plus so einem Wandler.  

Die Frage beim ESP32S3 Fusion Board ist, welche die DTU dafür verwendet. Da hilft wohl bloß durchprobieren, sobald der SDM120 in den Einstellungen der DTU eingeschaltet ist, müsste ja eine der Schnittstellen ein Lebenszeichen von sich geben.

Den internen RS485 Chip zu verwenden wäre zwar schön, allerdings scheint der anders angesteuert zu werden, da scheinbar RE und DE getrennt gesteuert werden.  15 RE, 16 R, 45 D, 46 DE,

Ich kenne nur die Version mit dem Wandler und da gibt es zwei Varianten.
Siehe auch: https://github.com/KlausLi/Esp-Soyosource-Controller

Eine verwendet RX und TX aber auch DE+RE , der ESP schaltet also von RX auf TX und umgekehrt. 

Die andere Variante, die ich verwende, hat einen IC drauf der die DE+RE Steuerung übernimmt und nur eine serielle schnittstelle mit RX und TX braucht. (Ich vermute das in dem IC ein FIFO speicher drinnen ist und ein bischen logik)

Ich fürchte, um den eingebauten RS485 verwenden zu können müsste wohl am Sourcecode der DTU geschraubt werden.

markusdd, 2024-06-11:
in openDTU on battery ist der RS485 vom Fusion voll unterstützt. korrekt.

Adminius, 2024-06-19:
aber nur wenn es um die Batterie geht, wenn es um den SDM Zähler geht, dann nicht :/

Solar320, 2024-06-19:
Normalerweise sollte das Bussystem es zulassen das sogar mehrere Geräte mit verschiedenen Adressen an einem Anschluß laufen.
Vieleicht gibts aber noch Probleme mit dem DTU seitigen Softwaresupport, 

Da müsste es dann etwas geben was "SDM" heist oder eher allgemein "RS485"  

"RS485-1": {
"rx": 16,
"rxen": 15,
"tx": 45,
"txen": 46,

und zusätzlich 

Battery "addr" 0x18,
SDM "addr" 0x15  als Beispiel....so in der Art. Kommt bestimmt noch irgendwann, wobei ich dann überall isolieren würde, man weis ja nie wo sich der Strom einen Weg bahnt. Wenn das BMS selber keine isolierte schnittstelle hat wirds schon wider schwieriger.

Adminius, 2024-06-20:
eben, für Batterie gibt es schon. Für SDM nicht. RS485 wäre zwar möglich, aber die Verwaltung wäre komplexer und isolieren macht in diesem Fall mehr als Sinn somit gleich 2x RS485 Transceiver und Problem ist gelöst.

iroc-fabi, 2024-07-01:
Ich hab jetzt mehrmals versucht die pin_mapping Datei für das Fusion Board anzupassen.
1:1 wie in der Beschreibung in eine Leere Datei kopiert. Jedesmal kommt:
Syntaxfehler in 'pin_mapping.json': Expected double-quoted property name in JSON at position 709 (line 33 column 9)

markusdd, 2024-07-01:
Dann teile doch mal deinen Versuch hier, das können wir sicher lösen.

iroc-fabi, 2024-07-02:
[pin_mapping - Victron.json](https://github.com/user-attachments/files/16061926/pin_mapping.-.Victron.json)

markusdd, 2024-07-02:
Komma zu viel bei tx2

markusdd, 2024-07-02:
in der GUI den Victron aktiviert?

RX/TX evtl. vertauscht?

iroc-fabi, 2024-07-02:
Hab ich bereits vertauscht. 
In der GUI hab ich bei Battery den SmartShunt ausgewählt. 
Muss zusätzlich bei VE.Direct das auch aktiviert werden? Hatte ich ausprobiert, keine Veränderung. 
woher weiß openDTU an welchen TX ich den angeschlossen habe?

markusdd, 2024-07-02:
der lauscht auf beiden kanälen.
es macht sinn das detail-logging anzuschalten.
und ja VE-direct muss an sein.

markusdd, 2024-07-02:
gut ist das definitiv nicht, weil du damit quasi -5V angelegt hast.

der Smartshunt sollte unbeschädigt sein weil der hat das ja als ausgang.

versuch wäre vllt mal den kanal 2 zu nutzen? Der ist ja völlig separat.

iroc-fabi, 2024-07-02:
Leider auch ohne Rückmeldung in openDTU.

markusdd, 2024-07-02:
hast du discord?

ich glaube dein Fall wäre da besser aufgehoben, da kann dir sicherlich jemand helfen.
Ich hab leider selbst noch ncihts mit dem shunt gemacht und kenne die nötigen configs nicht

da wären die experten im on battery support sicher hilfreicher

mastertape, 2024-07-30:
ganz doofe frage ich habe hier ein typ 3. display (OLED Display SH1106) Folgende anschlüsse vcc gnd sck sda - wo muss ich den sck anschließen? auf SCL wie beim oled anschluss vorgesehen scheint nicht zu funktionieren

markusdd, 2024-07-30:
scl und sck sind das gleiche (serial clock)

Wenn das nicht geht liegt das problem woanders.

closed 2024-08-25 by markusdd