markusdd/OpenDTUFusionDocs #12: wie schließe ich ein Display SSD1309 an?

Diggikomm, 2024-08-11:
Ich habe mir ein Display gekauft aber die Pin Bezeichnung unterscheidet sich von der auf der Fusion 2.2 Platine.

[https://www.amazon.de/dp/B0CL6WMHJN?psc=1&ref=ppx_yo2ov_dt_b_product_details](url)

Angeblich soll COMMUNICATION INTERFACE :4-wire SPI (default) / I2C sein.

Auch finde ich keine pin_mapping.json in der das Display aufgelistet ist.

Kann mir als Neuling jemand helfen

Danke für Eure Mühe

Sorbat, 2024-08-11:
Bei mir hat dein link nicht funktioniert, hab es mir über die ASIN rausgesucht. Hast du dieses Display? https://amzn.to/3YG1sfX

Diggikomm, 2024-08-11:
Ja genau das ist es, habe den Link nochmal aktualisiert.

Sorbat, 2024-08-11:
Ich hab ein paar verschiedene Displays, allerdings haben die alle nur 4 Pins. 
Bist du sicher, dass das mit diesem Display funktioniert?

Diggikomm, 2024-08-11:
Nein, ich bin nicht sicher. Ich hatte noch nicht die Erfahrung und habe einfach nach einem Display gesucht für opendtu.
Aber ich habe beiträge gefunden die wo Display mit SSD1309 aufgefürt wurden. Nur hatte der andere Probleme und anscheinen auch kein Fusion Board 

https://github.com/tbnobody/OpenDTU/discussions/1345

markusdd, 2024-08-11:
welche pins sind dir unklar?
das pinout ist für ein SPI-Display auf jeden fall richtig.

Diggikomm, 2024-08-11:
Ich habe Beispielsweise auch dieses Display gekauft.
https://www.amazon.de/dp/B0B7RPCZ4Z?psc=1&ref=ppx_yo2ov_dt_b_product_details

Da sind die Pins mit den Bezeichnungen die ich auch 1 zu1 auf dem Fusions Board finde, und es funktioniert sofort, bei der Auswahl der "OpenDTU Fusion v2 with CMT2300A, NRF24 and SSD1306 Display"

Und bei dem Board gibt es die Pinbezeichnungen RST, DC, CS, CLK, DIN, GND, VCC.

Das bedeutet das ich nicht weis wo die Anschlüsse DC, CS, CLK, DIN, hin kommen auf dem Fusions Board

Auserdem gibt es die Auswahl für SSD1309 in der pin_mapping.json nicht.

markusdd, 2024-08-11:
CS = CSN
DIN = MOSI
DC = DC (?)
CLK = SCK

Wie das display modell selbst richtig eingestellt wird oder ob dafür der SSD1306 trieber ok ist bitte auf dem community discord erfragen. Ich weiß nicht ob das 1309 nativ von OpenDTU softwareseitig supported wird.

Diggikomm, 2024-08-11:
Danke für deine Hilfe, ich werde dort mein Glück versuchen und fragen.

Ich check deine Auflistung leider nicht ganz, zur Sicherheit noch mal die unterschiedlichen Bezeichnungen welch ich zuordnen muss.

Fusion Board: GND, VCC, SCL, SDA
Display: RST, DC, CS, CLK, DIN, GND, VCC.

Welche Anschlüsse passen jetzt zusammen, denn das ist ja wahrscheinlich erst mal die Vorraussetzung bevor die Software-Seite geklärt werden kann, oder?

markusdd, 2024-08-11:
du guckst am völlig falschen header.
SCL/SDA ist i2c.

du musst am zentralen gucken, das ist der für SPI-Displays.

markusdd, 2024-08-11:
und falls das waveshare was du da gepostet hast beides (SPI und I2c) kann, dann musst du in deren dokumentation gucken wie der anschluss an i2c zu realisieren ist weil die pinnamen auf dem PCB deuten auf die SPI konfiguration hin.

Diggikomm, 2024-08-11:
Ich merke das ich mir da noch viel Wissen anlesen muss und ausprobieren nötig ist, denn das was in der Doku vom Display steht verstehe ich nur zu einem gerinen Prozentsatz :-(

https://www.waveshare.com/wiki/2.42inch_OLED_Module

Diggikomm, 2024-08-11:
Ich denke das der Abschnitt von bedeutung ist, nur check ich noch nicht was jetzt zu tun ist.

I2C:

    An der I2C-Position sind zwei 0R-Widerstände angeschlossen; DIN stellt eine Verbindung zum Host-SDA her und CLK stellt eine Verbindung zum Host-SCL her.
    Der DC-Pin kann zum Ändern der I2C-Adresse verwendet werden: Low einstellen, die 7-Bit-I2C-Adresse ist 0x3C; Bei hoher Einstellung ist die I2C-Adresse 0x3D.

Diggikomm, 2024-08-11:
Kann es sein das ich die zwei Wiederstände (also eigentlich sind es ja nur Brücken und keine Wiederstände, oder?)
einfach auf die Seite wo I2C steht gebrückt werden müssen und die Seite SPI getrennt werden muss?

markusdd, 2024-08-11:
ich glaube es gibt solche Displays, also ja, das kann sein.

Hier wäre wie gesagt der discord zu empfehlen weil da springen Leute rum die diese Hardware selbst verwenden und idR auch die entwickler die den Displaysupport treiberseitig bereitstellen.

Diggikomm, 2024-08-11:
Da hast du recht, abeer das Display kostet ja nicht die Welt und ich werde es einfach mal versuchen mit dem Umlöten.
Ich werde dann berichten, bestimmt interesiert das dann auch andere oder ist für die Hilfreich, wenn sie den Chat im Netz finden.

Diggikomm, 2024-08-11:
OK ich habe die Wiederstände (R0) entfernt und dann nach IC2 eine Brücke gelötet, jetzt geht das Display an, aber es sind nur Flimmernte Streifen zu sehen :-(

Diggikomm, 2024-08-13:
Also um das Abzuschließen, wie versprochen die Rückmeldung.
Ich habe mir ein neues Hailege 2,42" SSD1309 128x64 OLED Display Modul 2,42 Zoll 4 Pin OLED LCD Display Modul (IIC Blau) 
gekauft und im pin_mapping.json SSD1306 ausgewählt.

markusdd, 2024-08-25:
closing - resolved

closed 2024-08-25 by markusdd