markusdd/OpenDTUFusionDocs #13: Not booting after Factory Flash

Nailuj1, 2024-08-23:
Hello,
after buying a Fusionboard 2.2 i connected it to the power as shown in [HW_Setup](https://github.com/markusdd/OpenDTUFusionDocs/blob/main/HW_SETUP.md). I flashesd [opendtu-onbattery-generic_esp32s3_usb.factory.bin](https://github.com/helgeerbe/OpenDTU-OnBattery/releases/download/2024.08.18/opendtu-onbattery-generic_esp32s3_usb.factory.bin) without any error. 
``` 
``Using 'COM3' as serial port.
Connecting...
Detecting chip type... ESP32-S3
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-S3 (revision v0.2)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: NO
 - MAC Address: 80:65:99:C2:03:54
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
 - Flash Size: 16MB
Detected factory firmware Image, flashing without changes
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.4s
Flash will be erased from 0x00000000 to 0x001e5fff...
Flash params set to 0x024f
Compressed 1988176 bytes to 1250371...
Writing at 0x00000000... (1 %)
Writing at 0x000120d5... (2 %)
Writing at 0x00016d47... (3 %)
Writing at 0x0001ad04... (5 %)
Writing at 0x0001edc6... (6 %)
Writing at 0x00022dc1... (7 %)
Writing at 0x00026dda... (9 %)
Writing at 0x0002adee... (10 %)
Writing at 0x0002ede9... (11 %)
Writing at 0x00032de4... (12 %)
Writing at 0x00036ddf... (14 %)
Writing at 0x0003adda... (15 %)
Writing at 0x0003edd5... (16 %)
Writing at 0x00042dd0... (18 %)
Writing at 0x00046dcb... (19 %)
Writing at 0x0004e860... (20 %)
Writing at 0x000580e9... (22 %)
Writing at 0x00064736... (23 %)
Writing at 0x00070ace... (24 %)
Writing at 0x0007b90b... (25 %)
Writing at 0x00084606... (27 %)
Writing at 0x0008e49e... (28 %)
Writing at 0x000a1dfe... (29 %)
Writing at 0x000ae4ad... (31 %)
Writing at 0x000b47fc... (32 %)
Writing at 0x000bb926... (33 %)
Writing at 0x000c1fc7... (35 %)
Writing at 0x000cb10d... (36 %)
Writing at 0x000d1716... (37 %)
Writing at 0x000d8ab0... (38 %)
Writing at 0x000de57d... (40 %)
Writing at 0x000e49a5... (41 %)
Writing at 0x000eb694... (42 %)
Writing at 0x000f2c31... (44 %)
Writing at 0x000f945f... (45 %)
Writing at 0x000ffb58... (46 %)
Writing at 0x001059b3... (48 %)
Writing at 0x0010b629... (49 %)
Writing at 0x00111369... (50 %)
Writing at 0x00117761... (51 %)
Writing at 0x0011d7b5... (53 %)
Writing at 0x001234b4... (54 %)
Writing at 0x00129045... (55 %)
Writing at 0x0012e568... (57 %)
Writing at 0x0013408d... (58 %)
Writing at 0x001395ac... (59 %)
Writing at 0x0013e784... (61 %)
Writing at 0x001435d1... (62 %)
Writing at 0x0014867b... (63 %)
Writing at 0x0014d56a... (64 %)
Writing at 0x001525c8... (66 %)
Writing at 0x0015750a... (67 %)
Writing at 0x0015c647... (68 %)
Writing at 0x0016177c... (70 %)
Writing at 0x001666e5... (71 %)
Writing at 0x0016b6e9... (72 %)
Writing at 0x00170496... (74 %)
Writing at 0x00176052... (75 %)
Writing at 0x0017b740... (76 %)
Writing at 0x00180936... (77 %)
Writing at 0x00185b96... (79 %)
Writing at 0x0018a9fc... (80 %)
Writing at 0x0018fab0... (81 %)
Writing at 0x00194be3... (83 %)
Writing at 0x00199f53... (84 %)
Writing at 0x0019f489... (85 %)
Writing at 0x001a51f7... (87 %)
Writing at 0x001aa5f6... (88 %)
Writing at 0x001affd3... (89 %)
Writing at 0x001b5b38... (90 %)
Writing at 0x001bb16b... (92 %)
Writing at 0x001c37d5... (93 %)
Writing at 0x001ca0c1... (94 %)
Writing at 0x001d3079... (96 %)
Writing at 0x001d80fb... (97 %)
Writing at 0x001dd8eb... (98 %)
Writing at 0x001e3754... (100 %)
Wrote 1988176 bytes (1250371 compressed) at 0x00000000 in 12.6 seconds (effective 1261.7 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:

```
But the AP is not showing up in my WiFi.

I tried it with boot and also reset buttons but still not working.

Any suggestions?

Thanks in advance

Manos1966, 2024-08-23:
Ist es normal, dass derdual-core LX7 bis 240 MHz der ESP32-S3-WROOM-1U als:
>  - Chip Model: ESP32-S3 (revision v0.2)
>  - Number of Cores: 1
>  - Max CPU Frequency: 80MHz

erscheint? 🤔 

Oder ist alle ausgeschaltet und die ESP32-S3 laeuft mit dem sparsammeren Co-Prozessor?
(You can power off the CPU and make use of the low-power co-processor to constantly monitor the
peripherals for changes or crossing of thresholds)

Nailuj1, 2024-08-25:
I just waited some time and the AP was there in my Wifi. Did flash ower the [comunnity flasher](https://solar.metacontrol.eu/opendtu-onbattery-webflasher/).

closed 2024-08-25 by Nailuj1