markusdd/OpenDTUFusionDocs #15: Question: Does the  CAN/Iso extension shield fit into the AllianceApps Box?

Manos1966, 2024-11-03:
Sorry Markus, I did not know where to raise the question:

I see at the AllianceApps website, a Boxed version of the Fusion is offered:
https://shop.allianceapps.io/products/allianceapps-opendtu-fusion?pr_prod_strat=e5_desc&pr_rec_id=cc17aa39f

- Can I order this version (because "Laziness" is my middle name), open the box and add the CAN/Iso extension shield into it?
- Is there an elegant solution to get the CANbus RJ45 cable out of the AllianceApps box? (looking for ideas)

Best regards
Manos

markusdd, 2024-11-03:
The base Fusion DTU case is NOT built to support any of the Shields.

There is a community-made case though with support for the PoE shield that probably would be better suited: https://www.thingiverse.com/thing:6371201

Here you could use the hole for the RJ45 Jack to feed the cables inside.

Sorbat, 2024-11-21:
@Manos1966 vielleicht wäre das sonst eine alternative? https://3printd-solution.eu/Fusion-2x-Gehaeuse-Konfigurator-Extrudr-NX2-MATT-BLACK-RAL-9017-Fusion-Board-mit-PoE

Manos1966, 2024-11-21:
Danke @Sorbat 👍

markusdd, 2024-12-07:
I think this has been answered :D

closed 2024-12-07 by markusdd