markusdd/OpenDTUFusionDocs #22: Problem mit PoE und GND-Verbindung im Ethernetkabel

ingbrunner, 2025-01-03:
Ich bin auf ein Problem mit der PoE-Versorgung des Fusion Shields gestoßen:
Bei durchgehender GND-Verbindung des Ethernetkabels scheitert die PoE-Versorgung.

Hardware:
OpenDTU FUSION v2.3
OpenDTU FUSION SHIELD v1.1
NETGAER JGS524PE – 24-Port Gigabit Ethernet PoE Smart Managed Plus Switch 

Symptome:
- Mit durchgehender GND-Verbindung von Switch zum Fusion PoE Shield leuchtet D1 auf dem Fusion Board nicht (3,3V).
- Dafür wird U3 auf dem Fusion Shield warm (mit dem Finger spürt man unangenehme Wärme).
- Das Fusion Board ist offensichtlich nicht aktiv.
- Der Switch zeigt PoE-Versorgung an, aber keinen Link.
Dieses Verhalten wurde mit 2 verschiedenen Fusion Shields beobachtet: ich meinte bei ersten, ich hätte es "zerschossen", und habe ein zweites bestellt.

Lösung:
- Mit einem nicht abgeschirmten Patchkabel (ohne GND) oder mit dem in ersten Test verwendeten Kabel mit isoliertem Steckergehäuse (einfach mit Tesa-Band isoliert) funktioniert PoE einwandfrei:
- Das Fusion Board ist aktiv: das Web-Interface ist erreichbar.
- Der Netzwerkport auf den Fusion Shield zeigt einen Link und Aktivität.
- Der Switch zeigt PoE-Versorgung und einen Link (100MBit/s) an.
Mit der Unterbrechung der GND-Verbindung funktionieren beide Fusion Shields.

Der Switch ist an einer strukturierten Verkabelung angeschlossen, die geerdet ist. Die im Test verwendete Netzwerkdose wurde auf korrekte Reihenfolge der Adern getestet. Diese Dose wurde schon zum Setup diverser PoE-Geräte verwendet. Nie gab es Probleme, weder mit 1GBit/s Ethernet noch mit PoE. Die verwendeten Patchkabel wurden ebenfalls getestet.

Ich bin gern zu weitern Tests bereit. Vielen Dank im Voraus für Eure Mühe!

Ingolf

markusdd, 2025-01-03:
Ethernet ist dafür designed eine komplette isolierung durchzuführen. (daher sowohl für PoE als auch die Datenleitung die Transformer)

Wenn die Grounds absichtlich verbunden werden kann das sehr hässliche Seiteneffekte zur Folge haben.

Der Grund dafür ist, dass ja ein Netzwerkkabel auch aus einem anderen Gebäude mit einem völlig anderen Erdpotential kommen kann.

ich will mir jetzt aufgrund der textuellen Beschreibung kein Urteil erlauben, aber das Fusion ist nicht dafür designed in irgendwelche externen GND-systeme eingebunden zu werden. Das passiert über die USB-Versorgung auch nicht, USB-Netzteile sind vom externen GND-DC-seitig auch komplett isoliert.

Der Fehler in diesem Setup ist also das Fusion an einen externen GND zu koppeln, das geht aus Prinzip irgendwann schief.

In deinem Fall wirkt es mir fast, als ob da sogar ein reverse-biasing auftritt wenn U3 unangenehm warm wird. Das kann dann auch zur Zerstörung führen.

ingbrunner, 2025-01-03:
Richtig: bei Ethernet sind die Datenleitungen galvanisch getrennt. Der Shield/GND zwischen Geräten ist aber potentiell durchgängig. Die galvanische Trennung betrifft nicht die Gehäusemasse. Es gibt geschirmte Pachtkabel, die eine Masseverbindung herstellen. Und diese sind bei hohen Datenraten (>=1GBit/s) eigentlich Standard (bei kurzen Strecken funktioniert es auch ohne Shield).

Ein Switch ist grundsätzlich geerdet (von den Billiggeräten einmal abgesehen): zumindest über den PE, meist gibt es noch einen Extra-Erdanschluss. Das gleiche gilt auch für Patchfelder. Dann liegen die Gehäuse der Buchsen im Switch/Patchfeld auch auf Erdpotential. Und bei geschirmten Kabeln folglich auch das Endgerät. Also: die Signalleitungen sind sehr wohl galvanisch getrennt, aber über den Shield/GND kann man sich treffliche Ausgleichströme einfangen. Zumal, wenn das Endgerät auch geerdet ist.

Genau dieses Setup mit dem Shield/GND ist bei mir passiert. Und dem Fusion Shield sollte es egal sein, auf welchen Massepotential es liegt. Immerhin hat es keine anderweitige Masseverbindung. Ausgleichsströme werden da nicht fließen. Alle anderen Geräte funktionieren ja an meiner Verkabelung auch. Ich weiß, das ist kein Beweis, aber ein Hinweis, wo der Fehler vielleicht zu finden sein könnte.

Wenn größere Potentialdifferenzen in Gebäuden auftreten sollten, hilft es, den Shield/GND an einer Stelle zu unterbrechen. Ober aber man verwendet gleich Lichtwellenleiter (Glasfaser). Die Verwendung von Lichtwellenleitern ist nicht nur zwischen verschiedenen Gebäuden Standard, sondern auch zwischen den verschiedenen Etagen eines großen Gebäudes. Das ist natürlich bei kleineren Gebäuden etwas anderes: das Einfamilienhaus wird man mit Kat6-Kabel (oder besser) komplett von einem zentralen Punkt versorgen. Aber der Überspannungsschutz/Blitzschutz bedingt auch hier eine Erdung. In der Beschreibung jeder Netzwerkdose wird auf den korrekten Ausschuss des Shield/GND hingewiesen.

Viele Grüße

Ingolf

ingbrunner, 2025-01-05:
Ich habe die wahrscheinliche Ursache gefunden: es handelt sich um einen Designfehler des Fusion Shields.

Der  PoE-Regler auf dem Fusion Shield ist ein Si3404 von Skyworks Solutions. Man korrigiere mich bitte, falls ich die winzige Beschriftung falsch gelesen habe. Der prinzipielle Fehler würde auch bei einem anderen Schaltkreis existieren.
Anscheinend wurde das "Isolated Flyback Application Diagram" nach Figure 3.2 implementiert (https://www.skyworksinc.com/-/media/Skyworks/SL/documents/public/data-sheets/si3404-datasheet.pdf). 
Das ist letztlich auch nicht so wichtig, denn im Bereich des Spannungseingangs stimmen die vorgeschlagenen Designs nach den Abbildungen 3.1, 3.2 und 3.3 überein. Und hier liegt meiner Einschätzung nach der Fehler im Schaltungsdesign:
Das Signal "VNEG" in Figure 3.2 wurde offensichtlich mit dem "GND" (= Shield/Masse) der RJ45-Buchse für den Ethernet Eingang verbunden. Wenn man mit dem Multimeter den Widerstand zwischen C12 negativer Pol (in Figure 3.2: CDET) und der Abschirmung des RJ-45 Buchse misst, kann man das nachvollziehen (Achtung: Widerstandmessungen können Bauteile beschädigen). Das ist ein klassischer Designfehler, der jedem passieren kann.

Wenn man das Fusion Shield mit einem nicht abgeschirmten "billigen" Patchkabel versorgt, dann passiert nichts. Das dürfte bei den meisten Anwendern der Fall sein.

Die Folgen in einer Twisted-Pair Ethernet Installation mit durchgängig abgeschirmten Kabeln dagegen sind: über den Schirm der Kabel (Shield) wird "VNEG" auf "GND" gelegt, also auf Erdpotential oder Masse. In letzter Konsequenz landet "GND" via PE (Schutzleiter) auf der Potentialausgleichschiene des Gebäudes. Fest steht: diese Verbindung von "VNEG" zu "GND" ist so nicht vorgesehen.

Die elektrischen Folgen dieses Fehlers habe ich nur oberflächlich durchdacht. Kann sein, dass ich mich da irre. Ich halte folgendes Szenario für wahrscheinlich: "VNEG" wird auf "GND" gelegt (bei mir zum Glück mit dem Leitungswiderstand von ca. 10m Kabel). Folglich steht nur ca. die Hälfte der Spannungsdifferenz zwischen "VNEG" und "VPOS" zur Verfügung. Der Spannungsregler will nachregeln und er wird heiß. Die Spannung reicht nicht zum Betrieb des Fusion Boards aus.

Ich erlaube mir, 3 Varianten den Umgang mit diesen Designfehler vorzuschlagen:

1. Abhilfe im Falle der aktuellen Boardrevision (Fusion Shield 1.1):
- Billig-Patchkabel ohne Shield/GND verwenden (nur mit Plastikstecker ohne Metall)
- Wenn bessere Patchkabel mit Shield (GND) verwendet werden:
	- entweder den Stecker isolieren (Tesa-Band, Isolierband),
	- oder den Shield unterbrechen. Zum Vergleich, was ich mit dem Anschluss für den Shield meine: z.B. nach "Hirose Stecker RJ45" im WWW suchen.

2. Mittelfristige Abhilfe:
- auf dem Fusion Shield die Verbindung zwischen "VNEG" und dem "GND" der RJ45-Buchse unterbrechen, also das Board ändern.

3. Vogel-Strauß-Algorithmus (der ist bei einigen Betriebssystemen beliebt):
- hoffen, dass der Fehler bei wenigen Leuten aufritt und ignorieren.

Für mich ist die Lösung 1 die passende. Ich betrachte den Fall für mich als erledigt.

Viele Grüße

Ingolf

markusdd, 2025-01-05:
Hallo. 

VNEG ist nicht mit dem Shield verbunden, sondern VSS, was ein weiterer Ground auf der Ethernet-Seite ist. Habe das gerade auf einem PoE was ich hier als engineering-Referenz habe nachgemessen.
VNEG ist die negative Seite der Brückengleichrichterausgänge, von dort lässt es sich leicht messen.

Wenn du von VNEG zum Shield durchgang hast, ist definitiv etwas kaputt.

Zum PoE generell: Der Bedarf und die Verkaufszahlen sind sehr niedrig.
Wir werden das im Frühjahr, wenn der BWK-Bedarf üblicherweise wieder anzieht nochmal evaluieren, dass es aber nochmal einen re-spin dieses Boards geben wir ist derzeit zumindest einmal unwahrscheinlich.

markusdd, 2025-01-05:
Ich bin jetzt nochmal meine Referenzdokumente durchgegangen die ich für das Design herangezogen habe, sowohl öffentlich verfügabre 3404 Designs als auch deren Referenzboard:

Das Problem ist, und man entschuldige meinen Ausdruck hier: Der PoE-Standard ist eine Dorfmatratze an die jeder mal ran durfte.

Es gibt hier keine universelle Lösung.

Es gibt die Variante: Shield unconnected lassen. Es gibt auch die Variante: den digital GND nehmen
VNEG ist immer falsch, also bleibt noch VSS.

Problem: wenn ich unconnected mache, und jemand nicht ein setup wie du hat, wird das Shield eine riesige unreferenzierte Antenne.

Wenn ich digital GND nehme, komme ich auf dem Basis-Fusion Board in teufels Küche mit dem USB. Wenn jemand das Board an seinem Rechner stecken hat, und hat ein anderen Erdpotenzial, kann ich dem das Mainboard frittieren, oder sien USB-Netzteil.

Bleibt noch VSS.

Also die Antwort ist: Wie man es macht, macht man es verkehrt.

markusdd, 2025-01-05:
Um es nochmal im Detail auszuführen: Der PoE-Standard 'übersieht' die Möglichkeit, dass es Geräte geben kann, die Class-2 sind (also ohne Erdung), und trotzdem eine weitere Verbindung zu einem Gerät haben können (siehe Fusion Basisboard und USB).

Wenn man nur Class 2 hat, kann man den digital GND nehmen als Referenz für das Shield. Das wäre z.B. was ein Access-Point mit Plastikgehäuse tun würde.

Im Fusion kann ich auch das aus genannten Gründen nicht tun, weil ich die USB-Seite nicht ignorieren kann (außer ich hätte unendlich Geld für frittierte Notebooks oder Mainboards haha).

Es gibt also immer einen Fall wo eine Variante falsch ist.

Vorschlag wäre hier, in die Doku aufzunehmen, ein Kabel ohne Shiel-dConnection zu verwenden.

ingbrunner, 2025-01-06:
Hallo,
der Vorschlag, ein Kabel ohne Shield in die Dokumentation aufzunehmen, wäre meiner Meinung nach ausreichend. Das muss Du entscheiden.

Ich hatte anhand eines Shelly pro 3EM eine andere Idee: wie wäre es mit einer RJ-45 Buchse auf dem Fusion Shield, die insoliert ist? Zumindest ist sie das von innen, außen führt nach den Bildern ein Blechstreifen herum:
https://shelly-forum.com/thread/18815-teardown-shelly-pro-3em-stromaufnahme-und-speisung-mit-dc/
Oder hier bei Conrad eine Buchse komplett ohne Blech:
https://www.conrad.de/de/p/assmann-wsw-a-20142-modulare-einbaubuchse-a-20142-buchse-einbau-vertikal-rj45-pole-8p8c-schwarz-1-st-716232.html?hk=SEM&WT.mc_id=google_pla&utm_source=google&utm_medium=cpc&utm_campaign=DE+-+PMAX+-+Nonbrand+-+Connectors+%26+Cables&utm_id=17937931844&gad_source=1&gclid=CjwKCAiAm-67BhBlEiwAEVftNgLL8C-Udwg4N1cqKPqad7fbc-tuBd-qC9NkLLFwsHUl58h-hl8SgRoCZEIQAvD_BwE&refresh=true
Viele Grüße
Ingolf

ingbrunner, 2025-01-06:
Gern geschehen, ich wollte je verstehen, was da genau los ist.
Viel Erfolg weiterhin für das tolle Projekt!

markusdd, 2025-01-06:
closing with https://github.com/markusdd/OpenDTUFusionDocs/blob/main/POE.md#usage-in-setups-with-earthed-cable-shielding

closed 2025-01-06 by markusdd

xeno010, 2026-08-24:
@markusdd @ingbrunner 

Sehr interessantes Thema. Ich hab mir vor kurzem auch ein PoE Shield gekauft und auch die Hinweise in der Readme gesehen. 
Aus interessenhalber versuche ich das ganze gerade nachzuvollziehen und habe mir auch einige Referenz Design angeschaut.

Wäre nicht die beste Möglichkeit um alle Fälle abzudecken (UTP, STP (un- & geerdet)) Kabel, USB usw.:

Die Schirmung der ETH-Buche als separate "Schirm-GND" zusehen und diese HF technisch über Kondensatoren mit GND (nicht VSS) zu verbinden? 
Das hätte den Vorteil, dass DC - Ströme geblockt werden aber HF-Störungen abgeleitet werden können. 
Das sollte dann gleichermaßen bei UTP und STP (un- & geerdet) Kabel funktionieren (Auch wenn USB angeschlossen wurde).

TI hatte dazu ein Ref Design beschrieben: https://www.ti.com/lit/ug/tidu274/tidu274.pdf
-> 4,7nF + 1MOhm gegen GND. 

Was haltet ihr davon?

markusdd, 2026-08-24:
Hi. Tatsächlich macht das PoE Shield schon genau das:

[image]

[image]

Wir haben das Shield an VSS (was der 'Ground der PoE Seite ist, in TIs note verwirrenderweise als 'EARTH' bezeichnet, im text aber bei denen auch als GND, was falsch ist). Und dann haben wir einen 1nF Koppelikondensator rüber zu GND (was der Betriebsground des ganzen Fusions ist auch der der USB-Domäne bzw das 'Minus' des DC Power Supplies wenn eins verbunden ist)).

Der 1nF kommt einfach aus der Referenz von Silicon Labs deren Chip wir nutzen, TI hat halt 4.7nF, Prinzip ist das Gleiche.

Das Problem ist - und dafür muss man rauszoomen - in der Tatsache zu sehen, dass unklar ist, was 'EARTH' eigentlich sein soll.

Wann tritt das Problem auf: Wenn die Bezugserde von VSS (die gleich VSS ist bei einem gegroundeten LAN-Kabel,  weil das Shield ja da dran angeschlossen ist) ungleich der Bezugserde von GND ist.
Die Bezeugserde von GND sollte niemals bei isolierten Netzteilen direkt die lokale Erde sein, (USB Netzteile sind ja normalerweise isoliert, und DC Netzteile am Klemmblock auch), aber dieser GND ist über Y-Kondensatoren im Netzteil natürlich wegen EMI auch auf die echte Erde (den PE im Hausnetz) bezogen. D.h. der funktionale GND vom Fusion ist gekoppelt deine lokale Erde.

Nun kommt aber folgendes Problem: Bei Wasserleitungen und Gasleitungen wird vor allem in TT-Netzen (wo das Haus seinen eigenen Anlagenerder hat und nicht den vom Ortstrafo) ein sogenanntes Bonding an die lokale Erde vorgneommen, damit man nicht plötzlich dort Berührspannungen von über 50V bekommt lokal, wenn die reinkommenden Leeitungen draußen ganz woanders geerdet sind. Da gibt es teils sogar nicht-leitende isoliermuffen die sicherstellen, dass es von der lokalen Hausleitung keine elektrische Verbindung raus auf die Straße zurück zum Versorger gibt.
Warum erkläre ich das solange: Genau diese vorschrift fehlt bei Ethernet. Wenn ich jetzt zwischen 2 Häusern oder allg. elektrischen Anlagen, die ihren eigenen Betriebserder mit unterschiedlichen Bezugspotenzial haben ein geerdetes Ethernet Kabel ziehe, dann habe ich in meiner Kette von Koppelkondensatoren gegen EARTH plötzlich 2 unterschiedliche Bezugspotenziale.
Auf der einen seite zerre ich an meinem GND und VSS gegen die Erde die über das ETH Kabel kommt, auf der anderen zerrt einer an GND (und über die Kopplung auch an VSS) Richtung der lokalen Erde.

Nun kann man rätseln wer gewinnt, und im allgemeinen ist die Antwort unbekannt.
Elektrisch passieren tut erstmal nicht, weil ja kein Strom fließt, aber die Schaltung funktioniert nicht mehr sauber weil die verschiedenen GNDs hin und her schwimmen. (das ist ja das Symptom, PoE geht nicht).

Nun gibt es 3 Dinge das zu fixen:
1) Nicht an VSS sondern GND koppeln: klingt wie ne geile Idee, und würde gehen solange das PoE und Fusion nicht an ein USB Kabel oder lokalen Netzwerk angeschlossen sind (Weil dann 'floated' der GND ja ggü der echten Erde). Sobald aber einer da von seinem PC ein USB Kabel dransteeckt, kanns sein dass ich dem 50V und mehr PE-Spannnungsdifferenz in seinen Mainboard-USB-Controller jage, weil am Shield der 'remote ground' vom ETH Kabel anliegt und sein rechner aber als Nullpotenzial die lokale Erde sieht.
Perfektes Rezpoet um tausende € Hardware über USB zu vernichten. Also -> geht nicht.
2) Shield floaten lassen: Das geht technisch einwandfrei, ist aber eine EMI-Katastrophe. Damit dürfte ich kein Produkt verkaufen, weil dann ist meine Ethernet-Buchse und das angeschlossene Kabel eine riesige Antenne.
3) Und das ist die Lösung aus der FAQ: Ich mache es wie beim Wasser und Gas und sorge dafür, nicht die Erdpotenziale zweier verschiedener Anlagenerder über andere Infrastruktur ungefragt zu koppeln.
Das hat nämlich auch noch eine andere Implikation: Wenn man diese Verbindung herstellt, und es gibt irgendwann man einen Erdschluss und eine ausreichend niederohmige Verbindung, dann kann es sein dass diese eine Hausstromanlage ihren Fehlstrom über das ETH-Kabel schön in das andere Gebäude und seine Netzwerktechnik ableitet.
Also ist die Lösung: Keine Erdverbindung von Kabeln zulassen die woanders herkommen und potenziell ein anderes Erdpotenzial haben (und da reichen hier bei wenigen Volt DC wirklich ein paar Ohm um genug Berührspannung zu erzeugen dass es die GND-Referenz stört. da geht nix kaputt, aber es geht halt nimmer).

Die ganzen Application Notes gehen halt nicht davon aus, dass an der anderen Seite noch ein anderes 'EARTH' Potenzial existieren könnte.

Sorry für den Roman, aber ich wollte deiner fundierten Aussage und Nachforschung wenigstens ordentlich begegnen. Es ist ein mistiges Problem.

ingbrunner, 2026-08-24:
Ich habe aktuell nur wenig Zeit, ich kann nicht wirklich mit diskutieren.
Aber etwas habe ich bemerkt: auf vielen Designs ist die Gehäusemasse (Chassis Ground, liegt am RJ45-Shield an) von der allgemeinen Masse (Main Ground Layer) mit einem Hochspannungskondensator (HV Cap, Keramik, SMD, ich glaube 1-2 kV) abgetrennt. Ich meine, das ist z.B. beim Olimex ESP32-POE der Fall. Da sind ja die Schaltpläne offengelegt.
Ich entnehme das auch einer Application Note von TI (S. 20-21 Abschnitt 9.2 Board Layout Guidelines)
https://www.ti.com/lit/pdf/slva531
Vielleicht wäre das eine Option.

markusdd, 2026-08-24:
Siehe Bild oben, genau das machen wir. VSS und GND sind mit einem 1kV 1nF gekoppelt.