markusdd/OpenDTUFusionDocs #24: OpenDTU Fusion PoE configuration with Ethernet

kyberias, 2025-03-01:
I've recently acquired the combination of OpenDTU Fusion + PoE hat and now trying to get the Ethernet connection working on OpenDTU. The hat seems to power the device correctly with PoE but ethernet configuration seems to be missing, OpenDTU continues to use Wifi.

The default opendtu_fusion.json config file doesn't seem to enable Ethernet for any of the PoE options. 

How exactly should we configure OpenDTU to use Ethernet with Fusion + PoE?

markusdd, 2025-03-01:
Base openDTU had longstanding issues supporting Ethernet properly on ESP32-S3 due to SPI management. That is why it's not in the default file.

The W5500 Pins are described here: https://github.com/markusdd/OpenDTUFusionDocs?tab=readme-ov-file#pin-mapping-of-the-esp32-s3-on-opendtu-fusion-v2x 

An example mapping using these with Fusion is here in the OpenDTU repository:
https://github.com/tbnobody/OpenDTU/blob/a07a0275018856151a9bdf78b02401dac3105cc7/docs/DeviceProfiles/opendtu_fusion.json#L209

kyberias, 2025-03-01:
Looks like the W5500 pins are already configured in the file I use. Are these sufficient to enable Ethernet connection on Open DTU?

markusdd, 2025-03-01:
I am frankly not aware of the most recent changes in OpenDTU on the software side regarding Ethernet.
Do you have this particular PoE profile selected in the UI? Just loading this JSON is not enough, the correct profile also needs to be selected and then in network settings Ethernet also probably needs activation.

markusdd, 2025-03-01:
I just checked back with the guys on the discord server, I am also not seeing ethernet settings pop up when w5500 is configured.

As a workaround you could use openDTU on battery via the webflasher, or wait until I have a response.

kyberias, 2025-03-01:
I can wait. Do you refer to the OpenDTU-onBattery fork? I don't really need the battery stuff. I guess that supports the same inverters as OpenDTU?

markusdd, 2025-03-01:
openDTU on Battery essentially is the same as openDTU base firmware with extended support for battery communication, zero-feed etc.

you can operate it just with inverters though and use none of the battery stuff.

you can give it a try at https://solar.metacontrol.eu/opendtu-onbattery-webflasher/ (use chrome or edge) ans see if it works. for the fusion it shoudl automatically install the needed json already, no need to upload manually.

markusdd, 2025-03-01:
Found the problem!

please use this latest release binary and flash them via the web interface: 
https://github.com/tbnobody/OpenDTU/releases/download/v25.2.3/opendtu-generic_esp32s3_usb.bin 

then if you select the PoE profile ethernet will work!

[image]

[image]

markusdd, 2025-03-01:
feel free to re-open in case it does not work.

closed 2025-03-01 by markusdd

kyberias, 2025-03-01:
I tried what you suggested. While powered with USB and connected to Wifi, I was able to update to the latest binary and then upload the pin json file. After that I could select "OpenDTU Fusion v2 PoE" like this

[image]

After switching to PoE (with the PoE hat jumper) and powering via Ethernet cable, I can't connect to the board. It doesn't seem to acquire an IP address from DHCP. I have other PoE devices in the network, so it should work. The two leds (blue and red) light up and the PoE hat RJ-45 connector's green and amber lights both blink. Well, the green one blinks and the amber just stays on.

markusdd, 2025-03-01:
mmh that is more a question of local network setup, for me dhcp works fine.
maybe an address conflict? can you check your router/DHCP server if it finds the device?
Some of them get confused when the same dvice name appears with 2 different MAC addresses.

kyberias, 2025-03-01:
I was wondering is there still some setting in the json file or in the UI that I should set in order for Ethernet to be enabled in OpenDTU? I mean what is the mechanism of it switching to use Ethernet instead of Wifi?

markusdd, 2025-03-01:
when the w5500 is recognized ethernet is automatically enabled.

when a LAN cable is connected it will use ethernet, if not it will(should) fall back to wifi.

kyberias, 2025-03-01:
Sorry, I was confused about my network. My DHCP server placed it in another subnet. It now works as expected.

So I guess this was related to me first using an older build of OpenDTU. The latest build works.

Thanks for excellent support!

markusdd, 2025-03-01:
happy we solved it. enjoy your setup!

ogbrugge, 2025-04-15:
Thank you, this also worked for me!
Just a quick note to others looking here, the file from the first response in this thread by markusdd needs to be added as pin_mapping, not as main config or language pack in the Settings -> Config Management menu entry.