markusdd/OpenDTUFusionDocs #26: for clarification: no WLAN, no LAN?

traut21, 2025-04-16:
Just in order to fully understand that device:

* it does support both 2.4 and 0.9 GHz for HM/HMS/HMT hoymiles devices
* it does neither offer WLAN nor LAN
* only other interface is USB-C, not only for power, but also for data?

Sorbat, 2025-04-16:
> Just in order to fully understand that device:
> 
> * it does support both 2.4 and 0.9 GHz for HM/HMS/HMT hoymiles devices
> * it does neither offer WLAN nor LAN
> * only other interface is USB-C, not only for power, but also for data?

https://github.com/markusdd/OpenDTUFusionDocs/blob/main/pics/overview.png

traut21, 2025-04-16:
This image shows everything what it has, but it does not show what it does not have.

Obviously, there is no LAN port, compared to other openDTU devices which support e.g. LAN or even PowerLAN.

But there is no comment on WLAN either - and that's why I ask here. This image does not answer that question.

traut21, 2025-04-16:
Ah, after yet another careful inspection I now recognize 

**U.FL/IPEX 2.4 GHz Antenna Connector (ESP Wifi)**

I had not looked into that detail since I had expected that to be 2.4 GHz for HM, but there's yet another

**U-FL/IPEX 2.4 GHz Antenna Connector (Nordic RF)**

So it does in fact name a Wifi Antenna, where I have been looking for WLAN support.

closed 2025-04-16 by traut21