markusdd/OpenDTUFusionDocs #27: Nothing but red LED

boennemann, 2025-05-09:
It's the second unit that I bought now, I have no idea what's going on.

I ran the "AllianceApps OpenDTU Fusion" without any issues whatsoever for over a year.
And now all of a sudden after a few days the thing goes down, nothing but a red LED lights up, not reachable, nothing.

Sometimes I have success flashing via esptool and can start over, but now I'm at a point where a freshly bought unit just turned red after 3-4 days, I can successfully flash, even with erasing, but just nothing happens. The whole thing is dead.

```bash
esptool --port /dev/tty.usbmodem2101 erase_flash
Serial port /dev/tty.usbmodem2101
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 28:37:2f:ff:49:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.9s
Hard resetting via RTS pin...

esptool --baud 921600 --port /dev/tty.usbmodem2101 --chip esp32-s3 --before default_reset --after hard_reset write_flash 0x0 ~/Documents/opendtu-onbattery-generic_esp32s3_usb.factory.bin
esptool.py v4.8.1
Serial port /dev/tty.usbmodem2101
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 28:37:2f:ff:49:e0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00243fff...
Compressed 2375488 bytes to 1446827...
Wrote 2375488 bytes (1446827 compressed) at 0x00000000 in 11.9 seconds (effective 1598.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

How do I debug from here?

markusdd, 2025-05-09:
That certainly is not normal and also not common.

Can you elaborate a bit more about your setup?
- where is the DTU located
- what power supply do you use (we have seen issues with bad power supplies destroying boards or leading to unreliable operation - this would be my no. 1 point to check)
- did you always modify the software or did you run the device as distributed? 
- did you get a whole DTU or just the Community Edition board? - if just the board please specify which case and antennas or possibly other extension boards you use (anything, displays, shields etc.)

boennemann, 2025-05-09:
Hey @markusdd,

it's the following setup:

Hoymiles HMS-1000-2T Inverter
2 Victron Inverters with 4 modules each
4x Pylontech US3000C
Community Edition + ISO Can Shield

DTU is right next to the inverter.
Stable 2.4GHz Wifi is there and unchanged (and 5Ghz SSID is named differently)

Software is modified insofar as that I'm using opendtu-on-battery, not plain opendtu. I did this from the beginning.

Power supply is unchanged, but I can swap that one out.

Board revisions, I'll update you with.

markusdd, 2025-05-09:
What is your wifi antenna setup?

Also, please try a plain openDTU image, maybe even via the web flasher and not esptool and see if you get the wifi back. Also remove the shield for testing.

In such elaborate setups we need to check from simplest setup possible and then add stuff back in later.

If the power supply is unchanged and you have problems again please definitely try another one

markusdd, 2025-05-09:
also tagging @schlimmchen here, according to the logs above is that the correct way to flash a factory image of on battery? I have never used esptool for it on CLI because it was so bugged for S3 in the beginning.

markusdd, 2025-05-09:
and one hint about the LEDs: these will of course only work if the correct pin mapping is loaded.
So until this is done and wifi is setup, the other LEDs will do nothing.

closed 2025-07-21 by markusdd