markusdd/OpenDTUFusionDocs #28: Huawei 4850G2 on Fusion 2.3 not working ?!

killamilla0815, 2025-05-09:
Hi there

Is there anyone here wo managed to operate the Huawei 4850G2 PSU successfully (in combination with Pylontech battery, but the battery is already working via CAN, its just the Huaweit, that causes me headaches) on a Fusion 2.3 board ?
I went through all the documentation that is available for this topic and tried to set it up as "External CAN Controller with CAN transceiver" in combination with a MCP2515 CAB board (with 5V hack) but wasn't able to get it working.
I've tried both pin mappings, via the SPI header and also via the UART header, no sucess, there is always the same log output, no matter if the PSU is powered on or not:
22:48:24.562 > [Huawei::HwIfc] Failed to set parameter
22:48:24.562 > [Huawei::Controller] Cannot perform auto power control while critical PSU values are still unknown

Looking forward to learn from anybody, who managed to make it working on the Fusion board ideally in the same revision, how he managed this ?

Thanks

closed 2025-05-10 by markusdd

killamilla0815, 2025-05-10:
Hi @markusdd  

I was just looking for somebody around here, who is successfully running an OpenDTU Fusion v2.3 board with those PSU in order to tell me, if this can work at all. 
I've alrady replied to an open discussion in the OpenDTU/discussions section, but the guys there use all kinds of ESP boards, and I didnt found anyone who is operating a Fusion v2.3 under this constallation. Therefore, before spending more time on the troubleshooting I wanted to make sure, that this setup can even work at all, cause I've read all kind of stories about setups which cannot work due to xyz reasons.

Thanks.

Sorbat, 2025-05-10:
Hey, maybe take a look to this https://github.com/hoylabs/OpenDTU-OnBattery/discussions/1134
I don't know if you'll find any relevant information there. But this discussion is about the Fusion and the Huawei.